lvgl/lv_binding_micropython #272: Align to Micropython v1.20.0

amirgon, 2023-04-27:
Align to [Micropython v1.20.0](https://github.com/micropython/micropython/releases/tag/v1.20.0)

Related: https://github.com/lvgl/lv_binding_micropython/pull/242

sam0910, 2023-05-09:
Dear @amirgon ,
Is the Mp 1.20 will be aligned like before on lv_micropython?

Great thanks to you,

amirgon, 2023-05-09:
> Is the Mp 1.20 will be aligned like before on lv_micropython?

Yes, this is the plan.
Micropython v1.20 will be merged into lv_micropython master branch.

MichaelDu9226, 2023-05-29:
@amirgon We have a project plan to align to micropython 1.20.0, I would like to ask how is the progress of LVGL Align to Micropython v1.20.0
Thanks.

amirgon, 2023-05-29:
Hi @MichaelDu9226 ,
This is a high priority for me, the plan is to address this in the coming few weeks.  
As this involves many changes in all ports, I will need some help testing it on the different ports.

MichaelDu9226, 2023-05-29:
I'm familiar with esp32 and S3, I can help with testing if you need.

sam0910, 2023-06-01:
The browser start with 'lv_binding_micropython' for checking out the updates.
I look forward it. I believe many others do :)

amirgon, 2023-06-01:
> I look forward it. I believe many others do :)

I'm aware of this and sorry for the delay.

amirgon, 2023-06-05:
I made some progress.  
The unix port seems to be working correctly. Other ports still need some work.

See: https://github.com/lvgl/lv_micropython/pull/65

amirgon, 2023-06-18:
I've merged https://github.com/lvgl/lv_micropython/pull/65. Master branch is now aligned to Micropython v1.20.0
There's a known problem with the js/webassembly port that will be fixed.

I'll appreciate any help testing this on the different ports.  
Please test this and open new issues if you find any problems.

closed 2023-06-18 by amirgon