lvgl/lv_binding_micropython #277: SDCard is not working in STM32F769DISCO

ZXC-zxc, 2023-05-23:
Hello，I forked lv_micropython release/v8 branch, and complied the code to STM32F769DISCO board(by add a  C module display driver),it worked. But when I insert the SD card,the code cannot run ,it's stuck in the beginning. I tried the origin micropython with the same version（v1.19.1），it's worked fine，it mount the sdcard to "/sd" and can read the files.

What's the different between the two versions of micropython in sdcard , and what's next direction I should research , thanks.

amirgon, 2023-05-23:
@embeddedt - Looks like STM specific issue. any idea?

amirgon, 2023-05-24:
> I debug the code,I found the code crash at this line,when it check the "SKIPSD" file.https://github.com/lvgl/lv_micropython/blob/1b212e7c979c17c11aa9d61630ee41c68532fad1/ports/stm32/main.c#L547
> Is it something wrong with the vfs?

CC: @kisvegabor

kisvegabor, 2023-05-24:
Unfortunately I have no idea either :slightly_frowning_face: 
I've just tried out if the screen works well but I don't know much about its details.

ZXC-zxc, 2023-05-25:
Sorry bros,it's my fault.
In my display driver,I use dma2d to send the data to screen, but it's conflict with the dma handler in sdcard.
I disable the dam in my display driver ,it's worked as expect.

closed 2023-05-25 by amirgon