lvgl/lv_binding_micropython #282: Some compilers do not accept `.mp_obj_type` in 1.20

embeddedt, 2023-06-22:
I'm finding that Emscripten's clang version is refusing to accept the way some structs are being initialized now. This doesn't happen before the 1.20 merge so I assume it's related to upstream changing how object types are defined. Any ideas?

```
build/lvgl/lv_mpy.c:1138:20: error: initializer element is not a compile-time constant
    .mp_obj_type = mp_lv_LV_RES_type_base,
                   ^~~~~~~~~~~~~~~~~~~~~~
build/lvgl/lv_mpy.c:1184:20: error: initializer element is not a compile-time constant
    .mp_obj_type = mp_lv_ENUM_LV_LOG_LEVEL_type_base,
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
build/lvgl/lv_mpy.c:1246:20: error: initializer element is not a compile-time constant
    .mp_obj_type = mp_lv_LV_ALIGN_type_base,
                   ^~~~~~~~~~~~~~~~~~~~~~~~
build/lvgl/lv_mpy.c:1294:20: error: initializer element is not a compile-time constant
    .mp_obj_type = mp_lv_LV_DIR_type_base,
```

<details>
<summary>Code near line 1246</summary>

```c
STATIC MP_DEFINE_CONST_OBJ_TYPE(
    mp_lv_LV_ALIGN_type_base,
    MP_QSTR_LV_ALIGN,
    MP_TYPE_FLAG_NONE,
    print, LV_ALIGN_print,
    
    
    attr, call_parent_methods,
    
    
    locals_dict, &LV_ALIGN_locals_dict
);

STATIC const mp_lv_obj_type_t mp_lv_LV_ALIGN_type = {
#ifdef LV_OBJ_T
    .lv_obj_class = NULL,
#endif
    .mp_obj_type = mp_lv_LV_ALIGN_type_base,
};
```
</details>

embeddedt, 2023-06-22:
> Could you try to enable MICROPY_PY_THREAD in ports/webassembly/mpconfigport.h and see if you are getting the same error on upstream micropython?

I don't get the error when I do that.

> What is the clang version that you are using? Is it possible to switch clang version on Emscripten?

The previous version of Emscripten I was using included Clang 13. I'll update to the latest version (which uses Clang 17) and see what happens.

amirgon, 2023-06-22:
Could you try branch [fix/282](https://github.com/lvgl/lv_binding_micropython/tree/fix/282)?
I changed `mp_obj_type` to pointer, I think it should be anyway.

embeddedt, 2023-06-22:
Seems to have moved the problem to a different location. Progress?

```
build/lvgl/lv_mpy.c:15952:13: error: initializer element is not a compile-time constant
 15952 |     parent, mp_lv_obj_type.mp_obj_type,
       |             ^~~~~~~~~~~~~~~~~~~~~~~~~~
../../py/obj.h:783:250: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE'
  783 | #define MP_DEFINE_CONST_OBJ_TYPE(...) MP_DEFINE_CONST_OBJ_TYPE_EXPAND(MP_DEFINE_CONST_OBJ_TYPE_NARGS(__VA_ARGS__, _INV, 12, _INV, 11, _INV, 10, _INV, 9, _INV, 8, _INV, 7, _INV, 6, _INV, 5, _INV, 4, _INV, 3, _INV, 2, _INV, 1, _INV, 0)(mp_obj_type_t, __VA_ARGS__))
      |                                                                                                                                                                                                                                                          ^~~~~~~~~~~
../../py/obj.h:749:415: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE_NARGS_7'
  749 | #define MP_DEFINE_CONST_OBJ_TYPE_NARGS_7(_struct_type, _typename, _name, _flags, f1, v1, f2, v2, f3, v3, f4, v4, f5, v5, f6, v6, f7, v7) const _struct_type _typename = { .base = { &mp_type_type }, .name = _name, .flags = _flags, .slot_index_##f1 = 1, .slot_index_##f2 = 2, .slot_index_##f3 = 3, .slot_index_##f4 = 4, .slot_index_##f5 = 5, .slot_index_##f6 = 6, .slot_index_##f7 = 7, .slots = { v1, v2, v3, v4, v5, v6, v7, } }
      |                                                                                                                                                                                                                                                                                                                                                                                                                               ^~
../../py/obj.h:770:44: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE_EXPAND'
  770 | #define MP_DEFINE_CONST_OBJ_TYPE_EXPAND(x) x
      |                                            ^
```

```c
STATIC MP_DEFINE_CONST_OBJ_TYPE(
    mp_lv_canvas_type_base,
    MP_QSTR_canvas,
    MP_TYPE_FLAG_NONE,
    print, canvas_print,
    make_new, canvas_make_new,
    binary_op, mp_lv_obj_binary_op,
    attr, call_parent_methods,
    buffer, mp_lv_obj_get_buffer,
    parent, mp_lv_obj_type.mp_obj_type, // <---- this is the line that errors
    locals_dict, &canvas_locals_dict
);
```

amirgon, 2023-06-22:
> Seems to have moved the problem to a different location. Progress?

Maybe...  
I've updated the branch, could you try again?

embeddedt, 2023-06-22:
Unfortunately looks about the same to me.
```
build/lvgl/lv_mpy.c:15429:34: error: initializer element is not a compile-time constant
 15429 |     { MP_ROM_QSTR(MP_QSTR_CTRL), MP_ROM_PTR(mp_lv_LV_BTNMATRIX_CTRL_type.mp_obj_type) },
       |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../py/obj.h:348:23: note: expanded from macro 'MP_ROM_PTR'
  348 | #define MP_ROM_PTR(p) (p)
      |                       ^~~
build/lvgl/lv_mpy.c:15462:13: error: initializer element is not a compile-time constant
 15462 |     parent, mp_lv_obj_type.mp_obj_type,
       |             ^~~~~~~~~~~~~~~~~~~~~~~~~~
../../py/obj.h:783:250: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE'
  783 | #define MP_DEFINE_CONST_OBJ_TYPE(...) MP_DEFINE_CONST_OBJ_TYPE_EXPAND(MP_DEFINE_CONST_OBJ_TYPE_NARGS(__VA_ARGS__, _INV, 12, _INV, 11, _INV, 10, _INV, 9, _INV, 8, _INV, 7, _INV, 6, _INV, 5, _INV, 4, _INV, 3, _INV, 2, _INV, 1, _INV, 0)(mp_obj_type_t, __VA_ARGS__))
      |                                                                                                                                                                                                                                                          ^~~~~~~~~~~
../../py/obj.h:749:415: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE_NARGS_7'
  749 | #define MP_DEFINE_CONST_OBJ_TYPE_NARGS_7(_struct_type, _typename, _name, _flags, f1, v1, f2, v2, f3, v3, f4, v4, f5, v5, f6, v6, f7, v7) const _struct_type _typename = { .base = { &mp_type_type }, .name = _name, .flags = _flags, .slot_index_##f1 = 1, .slot_index_##f2 = 2, .slot_index_##f3 = 3, .slot_index_##f4 = 4, .slot_index_##f5 = 5, .slot_index_##f6 = 6, .slot_index_##f7 = 7, .slots = { v1, v2, v3, v4, v5, v6, v7, } }
      |                                                                                                                                                                                                                                                                                                                                                                                                                               ^~
../../py/obj.h:770:44: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE_EXPAND'
  770 | #define MP_DEFINE_CONST_OBJ_TYPE_EXPAND(x) x
      |                                            ^
build/lvgl/lv_mpy.c:15771:13: error: initializer element is not a compile-time constant
 15771 |     parent, mp_lv_obj_type.mp_obj_type,
       |             ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

amirgon, 2023-06-22:
> Unfortunately looks about the same to me.

Forgot to push, sorry.... Try now?

embeddedt, 2023-06-22:
Much better, now only the `MP_ROM_PTR` one remains.

```
build/lvgl/lv_mpy.c:12731:34: error: initializer element is not a compile-time constant
 12731 |     { MP_ROM_QSTR(MP_QSTR_FLAG), MP_ROM_PTR(mp_lv_LV_OBJ_FLAG_type.mp_obj_type) },
       |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../py/obj.h:348:23: note: expanded from macro 'MP_ROM_PTR'
  348 | #define MP_ROM_PTR(p) (p)
      |                       ^~~
```

amirgon, 2023-06-22:
> Much better, now only the `MP_ROM_PTR` one remains.

Fixed now?

embeddedt, 2023-06-22:
Seems to have carried on compiling now! Thank you!

amirgon, 2023-06-22:
When it's all good - we can merge https://github.com/lvgl/lv_binding_micropython/pull/284

commit 8d5362b7 (2023-06-22): Change mp_obj_type to pointer

Fixes #282
commit c457fb1b (2023-06-22): Change mp_obj_type to pointer

Fixes #282
commit 54614ecb (2023-06-22): Change mp_obj_type to pointer

Fixes #282
PR #284 (2023-06-22): Change mp_obj_type to pointer
Fixes #282
closed 2023-06-22 by embeddedt