lvgl/lv_binding_micropython #283: Try build for esp32_psram

straga, 2023-06-22:
```
[100/100] cd /opt/upy/lv_micropython/ports/esp32/build-GENERIC_SPIRAM/bootloader/esp-idf/esptool_py && /root/.espressif/python_env/idf4.4_py3.6_env/bin/python /opt/upy/tools/esp-idf-v4.4.5/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /opt/upy/lv_micropython/ports/esp32/build-GENERIC_SPIRAM/bootloader/bootloader.bin
Bootloader binary size 0x55f0 bytes. 0x1a10 bytes (23%) free.
[1075/1615] Generating ../../lv_mp.c
FAILED: lv_mp.c
cd /opt/upy/lv_micropython/ports/esp32/build-GENERIC_SPIRAM/esp-idf/main && /root/.espressif/python_env/idf4.4_py3.6_env/bin/python3.6 /opt/upy/lv_micropython/lib/lv_bindings/gen/gen_mpy.py -M lvgl -MP lv -MD /opt/upy/lv_micropython/ports/esp32/build-GENERIC_SPIRAM/lv_mp.c.json -E /opt/upy/lv_micropython/ports/esp32/build-GENERIC_SPIRAM/lv_mp.c.pp /opt/upy/lv_micropython/lib/lv_bindings/lvgl/lvgl.h > /opt/upy/lv_micropython/ports/esp32/build-GENERIC_SPIRAM/lv_mp.c || ( rm -f /opt/upy/lv_micropython/ports/esp32/build-GENERIC_SPIRAM/lv_mp.c && /bin/false )
  File "/opt/upy/lv_micropython/lib/lv_bindings/gen/gen_mpy.py", line 1769
    return (user_data if user_data_found else None), *get_user_data_accessors(containing_struct, containing_struct_name)
                                                     ^
SyntaxError: invalid syntax
[1089/1615] Generating /opt/upy/lv_micropython/mpy-cross/build/mpy-cross
make[1]: Entering directory '/opt/upy/lv_micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/mpversion.h
CC ../py/modsys.c
CC main.c
LINK build/mpy-cross
   text    data     bss     dec     hex filename
 305494   13848     840  320182   4e2b6 build/mpy-cross
make[1]: Leaving directory '/opt/upy/lv_micropython/mpy-cross'
[1091/1615] Building C object esp-idf/libsodium/CMakeFiles/...m/crypto_generichash/blake2b/ref/blake2b-compress-ref.c.ob
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [Makefile:67: deploy] Error 2
make: Leaving directory '/opt/upy/lv_micropython/ports/esp32'
```

amirgon, 2023-06-22:
Please provide a **complete** description how to **reproduce** your problem:
- Your system configuration, distribution etc.
- ESP IDF version
- lv_micropython version you are using
- The **exact** and **complete** list of commands that brought you to this error.
- Any other information that is useful for someone else to reproduce this error.

straga, 2023-06-23:
1. python:3.6-buster - docker
2. source /opt/upy/tools/esp-idf-v4.4.5/export.sh
3. https://github.com/lvgl/lv_micropython/commit/3f25a765dafd5742639b85e94e845e370f73057f
4. make -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16" BOARD=STRAGA_CORE_SPIRAM_16mb all
5. just run command from readme from lv_micropython

amirgon, 2023-06-30:
> esp32+ili9341+xpt2046 = work, but render very slow.

Do you use the [generic driver](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/generic/ili9xxx.py) or the [ESP32 driver](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/ili9XXX.py)?

straga, 2023-07-02:
[ESP32 driver](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/ili9XXX.py)

from ili9XXX import ili9341

amirgon, 2023-07-02:
What is the frame rate you are seeing?  (you can enable `LV_USE_PERF_MONITOR` to see frame rate)
I've seen anything between 10FPS to 50FPS depending on the GUI complexity and the size of the area that is refreshed (LVGL tries to update only the areas that change between frames).  
Also, you didn't specify what you tried to run. Does it happen with any LVGL example, or only with your code?

straga, 2023-07-02:
Today rebuild all. Erase all. Also I am experement with gc.threshold(2000) that was problem. Now works right.

closed 2023-07-02 by straga