lvgl/lv_binding_micropython #285: Can't show Image with jpg format.

HonestQiao, 2023-06-28:
My device: ESP32-S3-Box-Lite (N16R8V)
DIsplay: ST7789

My code:
```
import lvgl as lv
import lv_utils
import fs_driver

lv.init()
fs_drv = lv.fs_drv_t()
fs_driver.fs_register(fs_drv, 'S')

/* config for display */

scr = lv.scr_act()
cover_img = lv.img(scr)
cover_img.set_src("S:/images/zt_cover.png")
cover_img.set_size(180, 240)
cover_img.align(lv.ALIGN.TOP_LEFT, 0, 0)
```

If I use "S:/images/zt_cover.png" for cover_img.set_src(), the image will be displayed.
[image]

If I use "S:/images/zt_cover.jpg" for cover_img.set_src(), the image wont be displayed. No error, only empty.
zt_cover.jpg is convert from zt_cover.png.
[image]

I tried to use ./lib/lv_bindings/lvgl/scripts/jpg_to_sjpg.py to convert zt_cover.jpg to zt_cover.sjpg, but zt_cover.sjpg can't be displayed either.

In ./lib/lv_bindings/lv_conf.h, LV_USE_SJPG is open:
```
/*PNG decoder library*/
#define LV_USE_PNG 1

/*BMP decoder library*/
#define LV_USE_BMP 1

/* JPG + split JPG decoder library.
 * Split JPG is a custom format optimized for embedded systems. */
#define LV_USE_SJPG 1

/*GIF decoder library*/
#define LV_USE_GIF 1
```

What is the problem?

amirgon, 2023-08-03:
I'm not sure it's related but ESP32S3 is not officially supported by lv_micropython.
Did you try to reproduce this problem in C? It's not clear to me if it's Micropython bindings issue of LVGL issue

PGNetHun, 2023-08-13:
Hello!
There is a bug in LVGL `lv_sjpg.c` file, the problem is the color format:
https://github.com/lvgl/lvgl/issues/4407#issuecomment-1655861234