lvgl/lv_binding_micropython #288: Input device KEYPAD can't work in micropython

jd3096-mpy, 2023-08-01:
We can easily find some examples for touchscreen,so it is easy to learn how to write KEYPAD drivers.
Here is an example:
`class POINTER:
    def __init__(self):
        if not lv.is_initialized():
            lv.init()
        disp = lv.disp_t.__cast__(None)
        self.ok=Pin(7,Pin.IN,Pin.PULL_UP)
        indev_drv = lv.indev_drv_t()
        indev_drv.init()
        indev_drv.type = lv.INDEV_TYPE.POINTER
        indev_drv.read_cb = self.read
        indev_drv.register()

    def read(self, indev_drv, data) -> int:
        data.point.x=10
        data.point.y=20
        if not self.ok.value():
            print('press enter')
            data.state=lv.INDEV_STATE.PRESSED
            return False
        else:
            data.state=lv.INDEV_STATE.RELEASED
            return False`
So I press the button GPIO7 to click the (10,20) ,it works.
But when I tried to use KEYPAD,like this:
`class KEYPAD:
    def __init__(self):
        if not lv.is_initialized():
            lv.init()
        disp = lv.disp_t.__cast__(None)
        self.ok=Pin(6,Pin.IN,Pin.PULL_UP)
        indev_drv = lv.indev_drv_t()
        indev_drv.init()
        indev_drv.type = lv.INDEV_TYPE.KEYPAD
        indev_drv.read_cb = self.read
        indev_drv.register()

    def read(self, indev_drv, data) -> int:
        data.key=lv.KEY.ENTER
        if not self.ok.value():
            print('press next')
            data.state=lv.INDEV_STATE.PRESSED
            return False
        else:
            data.state=lv.INDEV_STATE.RELEASED
            return False`

It doesn't work,what is the problem?
By the way,how to input letter 'a' use gpio change? I want to use blackberry q10 keyboard?
Can you give me a micropython KEYPAD examples? Thanks.

amirgon, 2023-08-03:
Does it work in C?  
Is it related to the Micropython bindings or a general LVGL problem?

jd3096-mpy, 2023-08-03:
> Does it work in C? Is it related to the Micropython bindings or a general LVGL problem?

I am sure it work in c
Maybe I just need the correct way to use it in micropython

closed 2023-08-03 by jd3096-mpy