lvgl/lv_binding_micropython #290: KeyError: lv_disp_drv_t_flush_cb

lbuque, 2023-09-17:
I want to use lvgl in rgb panel.

lv_micropython: master
lv_binding: https://github.com/imliubo/lv_binding_micropython/tree/dev_esp32_s3
esp-idf: v4.4.1

```python
import lvgl as lv
import tft_config
# import lv_utils

tft = tft_config.config()

def disp_drv_flush_cb(disp_drv, area, color_p):
    print("hahah")
    w = (area.x2-area.x1+1)
    h = (area.y2-area.y1+1)
    tft.bitmap(
        area.x1,
        area.y1,
        area.x2,
        area.y2,
        disp_drv.draw_buf.buf_act.__dereference__(2*w*h)
    )
    disp_drv.flush_ready()

lv.init()
# if not lv_utils.event_loop.is_running():
#     event_loop=lv_utils.event_loop()
#     print(event_loop.is_running())

# create a display 0 buffer
disp_buf0 = lv.disp_draw_buf_t()
buf1_0 = bytearray(480 * 10)
disp_buf0.init(buf1_0, None, len(buf1_0) // lv.color_t.__SIZE__)

# register display 0 driver
disp_drv_0 = lv.disp_drv_t()
disp_drv_0.init()
disp_drv_0.draw_buf = disp_buf0
disp_drv_0.flush_cb = disp_drv_flush_cb
disp_drv_0.hor_res = 480
disp_drv_0.ver_res = 272
disp_drv_0.user_data = {"display_index": 0}
disp = disp_drv_0.register()

lv.disp_t.set_default(disp)

scr0 = lv.obj()
# create button widget on screen 0
btn0 = lv.btn(scr0)
btn0.align(lv.ALIGN.CENTER, 0, -50)
label0 = lv.label(btn0)
label0.set_text("LVGL Screen 0")
lv.scr_load(scr0)

while True:
    lv.task_handler()

```

Running the above code, I get the following error:
```
Traceback (most recent call last):
  File "<stdin>", line 51, in <module>
KeyError: lv_disp_drv_t_flush_cb
```

Did I do something wrong there?

amirgon, 2023-09-17:
> Did I do something wrong there?

Remove this line:
```disp_drv_0.user_data = {"display_index": 0}`

PGNetHun, 2024-01-12:
@lbuque 
do you still have question/problem regarding this, or can I close the ticket?