lvgl/lv_binding_micropython #291: a bug in release/v8

lbuque, 2023-09-23:
I compile lv_micropython, and execute the following code:

```python
import lvgl as lv
import time
from espidf import SPI3_HOST
# from ili9XXX import ili9341
from xpt2046 import xpt2046
import tft_config

WIDTH = 480
HEIGHT = 272

tft = tft_config.config()

# create a display 0 buffer
disp_buf0 = lv.disp_draw_buf_t()
buf1_0 = bytearray(WIDTH * 10)
disp_buf0.init(buf1_0, None, len(buf1_0) // lv.color_t.__SIZE__)

# register display 0 driver
disp_drv = lv.disp_drv_t()
disp_drv.init()
disp_drv.draw_buf = disp_buf0
disp_drv.flush_cb = tft.flush
disp_drv.hor_res = WIDTH
disp_drv.ver_res = HEIGHT
# disp_drv_0.user_data = {"display_index": 0}
disp0 = disp_drv.register()
lv.disp_t.set_default(disp0)

scr0 = lv.obj()
# create button widget on screen 0
btn0 = lv.btn(scr0)
btn0.align(lv.ALIGN.CENTER, 0, -50)
label0 = lv.label(btn0)
label0.set_text("LVGL Screen 0")
lv.scr_load(scr0)

import time
while True:
    lv.task_handler()
    time.sleep(1)

```

The following exception occurred:

```shell
xtensa-esp32s3-elf-addr2line -e build-ELECROW_43/micropython.elf Backtrace: 0x42069d87:0x3fcb2620 0x4206cc07:0x3fcb2640 0x4206928c:0x3fcb2670 0x42068bbb:0x3fcb2690 0x420876bf:0x3fcb26b0 0x4203d0b5:0x3fcb26e0 0x4203ce89:0x3fcb2700 0x4201dcf9:0x3fcb2720 0x4201de0a:0x3fcb2740 0x40379bfd:0x3fcb2760 0x42018448:0x3fcb2800 0x4201dcf9:0x3fcb2860 0x4201dd22:0x3fcb2880 0x4206189b:0x3fcb28a0 0x42061a99:0x3fcb2930 0x4202b272:0x3fcb2980
??:0
/Volumes/NV2/Workspace/lv_micropython-v8/lib/lv_bindings/lvgl/src/core/lv_obj_pos.c:297
/Volumes/NV2/Workspace/lv_micropython-v8/lib/lv_bindings/lvgl/src/core/lv_obj_style.c:188
/Volumes/NV2/Workspace/lv_micropython-v8/lib/lv_bindings/lvgl/src/core/lv_obj_class.c:113
/Volumes/NV2/Workspace/lv_micropython-v8/lib/lv_bindings/lvgl/src/core/lv_obj.c:217
/Volumes/NV2/Workspace/lv_micropython-v8/lib/lv_bindings/lvgl/src/hal/lv_hal_disp.c:222
/Volumes/NV2/Workspace/lv_micropython-v8/ports/esp32/build-ELECROW_43/lv_mp.c:27649
/Volumes/NV2/Workspace/lv_micropython-v8/ports/esp32/build-ELECROW_43/lv_mp.c:103
/Volumes/NV2/Workspace/lv_micropython-v8/py/runtime.c:684
/Volumes/NV2/Workspace/lv_micropython-v8/py/runtime.c:700
/Volumes/NV2/Workspace/lv_micropython-v8/py/vm.c:1026
/Volumes/NV2/Workspace/lv_micropython-v8/py/objfun.c:288
/Volumes/NV2/Workspace/lv_micropython-v8/py/runtime.c:684
/Volumes/NV2/Workspace/lv_micropython-v8/py/runtime.c:658
/Volumes/NV2/Workspace/lv_micropython-v8/shared/runtime/pyexec.c:120
/Volumes/NV2/Workspace/lv_micropython-v8/shared/runtime/pyexec.c:277
/Volumes/NV2/Workspace/lv_micropython-v8/ports/esp32/main.c:176
```

what should I do? I want to use lvgl 8 on micropython

embeddedt, 2023-09-23:
I could be wrong but at first glance it appears you're missing a call to `lv.init()` at the beginning of the program.

lbuque, 2023-09-24:
@embeddedt You are right, I made a very stupid mistake.

closed 2023-09-24 by lbuque