lvgl/lv_binding_micropython #313: Display buffer initialization for ESP32

PGNetHun, 2024-01-10:
Hello @XuNeo, @kisvegabor ,

There is a buffer allocation issue in display initialization for ESP32 ili9xxx driver:
https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/ili9XXX.py#L151

These 2 lines allocate memory which can be accessed by DMA, and returns `void *`, so a simple memory area, and it cannot be passed to [lv_display_set_draw_buffers(lv_display_t * disp, lv_draw_buf_t * buf1, lv_draw_buf_t * buf2)](https://github.com/XuNeo/lvgl/blob/master/src/display/lv_display.c#L389), which was refactored in [this PR](https://github.com/lvgl/lvgl/pull/5204).

Why do we need `lv_draw_buf_t` **everywhere**, even for display buffer (and decoded image data)? Only to have memory-aligned data?

I see that there is also a [lv_draw_buf_init()](https://github.com/XuNeo/lvgl/blob/master/src/draw/lv_draw_buf.c#L140) function, which accepts a `void*` memory area, but it tries to align it.

Could you please check the above mentioned codes and help in this topic?

PGNetHun, 2024-01-11:
Problem **solved** in other [PR](https://github.com/lvgl/lv_binding_micropython/pull/314), and is merged to this `multi-instance` branch for [multi-instance PR](https://github.com/lvgl/lv_binding_micropython/pull/311).

Tested on real devices with ST7789 and ILI9342C displays.

closed 2024-01-11 by PGNetHun

kisvegabor, 2024-01-11:
I've raised some concerns about if we really need `draw_buf` in `lv_display_set_draw_buffers()`. Could you tell your opinion in https://github.com/lvgl/lvgl/pull/5204?

XuNeo, 2024-01-11:
You are doing it right to use lv_draw_buf_init.

> I see that there is also a [lv_draw_buf_init()](https://github.com/XuNeo/lvgl/blob/master/src/draw/lv_draw_buf.c#L140) function, which accepts a void* memory area, but it tries to align it.

The issue is you have configured to do so. 

https://github.com/lvgl/lv_binding_micropython/pull/314/files#diff-136397af91c41d3e65b59f16b7ae542b5545b37c353bd10417bdef6075be5821R82-R83

XuNeo, 2024-01-11:
> I've raised some concerns about if we really need `draw_buf` in `lv_display_set_draw_buffers()`. Could you tell your opinion in [lvgl/lvgl#5204](https://github.com/lvgl/lvgl/pull/5204)?

Let's open an new issue for it, so everyone can see it. Doing it now.