lvgl/lv_binding_micropython #317: problem building SAMD/SEEED_WIO_TERMINAL

nik1977, 2024-01-16:
I got this error. I don't know if I missed something.

LVGL-GEN build-SEEED_WIO_TERMINAL/lvgl/lv_mpy.c
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 1778
    return (user_data if user_data_found else None), *get_user_data_accessors(containing_struct, containing_struct_name)
                                                     ^
SyntaxError: invalid syntax
make: *** [../../extmod/extmod.mk:387: build-SEEED_WIO_TERMINAL/lvgl/lv_mpy.c] Error 1
make: *** Deleting file 'build-SEEED_WIO_TERMINAL/lvgl/lv_mpy.c'

PGNetHun, 2024-01-19:
Hello!
From which branch do you build it?

nik1977, 2024-01-19:
Hello. I used master branch. All actions on Raspberry Pi 4  bullseye .