lvgl/lv_binding_micropython #322: Abnormal color display on LCD with RP pico

yusuhua, 2024-01-27:
Recently, I used the latest master to compile the firmware of RP pico. When I ran examples/generic-st77xx-with-xpt2046.py, I found an error. I changed line 416 of st77xx.py from lv.disp_create to lv.display_create and it ran successfully, but It is found that the color display is abnormal, as shown in the figure:
[image]
The white background turned to light green, and the buttons turned to yellow.
After adding the bgr=True parameter, the button is blue, but the white background is still light green, and the text looks blurry.
[image]
The device I'm using is: https://www.waveshare.com/Pico-ResTouch-LCD-2.8.htm

PGNetHun, 2024-01-27:
Hello!
That's a known issue: some displays require that the 16 bit RGB565 pixel data bytes are swapped.
There is a function in LVGL that swaps the bytes: `lv.draw_sw_rgb565_swap`

I have fixed this issue in PR: https://github.com/lvgl/lv_micropython/pull/73
It is merged to main, so please pull repo, build, and test again.
Set `bgr` parameter to `False` to swap RGB565 bytes.

yusuhua, 2024-01-30:
I pulled the repo again and rebuilt a firmware, but found an error when running Thonny:
`>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (thonny.plugins.micropython.mp_back.ProtocolError: Unexpected read during raw paste).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
KeyboardInterrupt: 

MPY: soft reboot
MicroPython v1.20.0-703-gcc50c7aa5 on 2024-01-30; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> `

The display screen shows snowflakes:
[image]

Here is the Thonny's backend.log
[backend.log](https://github.com/lvgl/lv_binding_micropython/files/14100161/backend.log)

PGNetHun, 2024-01-30:
Please send your MicroPython code (I'm interested in display creation, and whether LVGL loop is initiated, and screenshot of serial port output (console output) when you restart. Attach these to the new comment.
So I can test the same code (but different display).

PGNetHun, 2024-02-04:
Hello @yusuhua ,
There was a bug in `st77xx.py` file, that variables `w` and `h` were used before set.
That is fixed, and I have tested it on Raspberry Pi Pico W + ILI9341 display, and it works.
Could you please pull repo and test again?

Note, that Pico has not so much memory, so increase `factor` parameter (split screen into smaller pieces for partial rendering), and disable double buffering to save more RAM.

This is how I connected ILI9341 display to Pico (without touch part):
[image]

Code:
```
import lvgl as lv

try:
    if not lv.is_initialized():
        print("Init LVGL")
        lv.init()
        print("Init LVGL - DONE")

    import lv_utils
    if lv_utils.event_loop.is_running():
        print("Deinit lv_utils.event_loop")
        lv_utils.event_loop.current_instance.deinit()
        print("Deinit lv_utils.event_loop - DONE")

    print("Create SPI")
    import machine
    spi = machine.SPI(
        0,
        baudrate=24_000_000,
        sck=machine.Pin(6, machine.Pin.OUT),
        mosi=machine.Pin(7, machine.Pin.OUT)
    )
    print("Create SPI - DONE")

    print("Init Ili9341")
    import ili9xxx
    drv = ili9xxx.Ili9341(rot=3, spi=spi, cs=13, dc=15,
                          rst=14, factor=8, doublebuffer=False)
    print("Init Ili9341 - DONE")

    print("Show screen")
    screen = lv.obj()
    button = lv.button(screen)
    button.center()
    label = lv.label(button)
    label.set_text("Hello World!")
    label.center()
    lv.screen_load(screen)
    print("Show screen - DONE")

except Exception as e:
    print(e)
```

closed 2024-02-05 by PGNetHun