lvgl/lv_binding_micropython #325: ESP-IDF V4.4 make errors

kk8900, 2024-02-01:
make -j -C ports/esp32 BOARD=GENERIC_S3_SPIRAM USER_C_MODULES=~/esp/lv_micropython/lib/lv_bindings/bindings.cmake

The error message is as follows,How to solve it?

CMake Error at /home/le/esp/lv_micropython/lib/lv_bindings/bindings.cmake:6 (include):
  include could not find requested file:

    /home/le/esp/lv_micropython/lib/lv_bindings/lvgl/CMakeLists.txt
Call Stack (most recent call first):
  /home/le/esp/lv_micropython/py/usermod.cmake:42 (include)
  main/CMakeLists.txt:14 (include)

CMake Error at /home/le/esp/lv_micropython/lib/lv_bindings/mkrules.cmake:39 (target_compile_options):
  target_compile_options called with incorrect number of arguments
Call Stack (most recent call first):
  /home/le/esp/lv_micropython/lib/lv_bindings/mkrules.cmake:98 (lv_bindings)
  /home/le/esp/lv_micropython/lib/lv_bindings/bindings.cmake:20 (all_lv_bindings)
  /home/le/esp/lv_micropython/py/usermod.cmake:42 (include)
  main/CMakeLists.txt:14 (include)

CMake Error at /home/le/esp/lv_micropython/lib/lv_bindings/mkrules.cmake:39 (target_compile_options):
  target_compile_options called with incorrect number of arguments
Call Stack (most recent call first):
  /home/le/esp/lv_micropython/lib/lv_bindings/mkrules.cmake:131 (lv_bindings)
  /home/le/esp/lv_micropython/lib/lv_bindings/bindings.cmake:20 (all_lv_bindings)
  /home/le/esp/lv_micropython/py/usermod.cmake:42 (include)
  main/CMakeLists.txt:14 (include)

CMake Error at /home/le/esp/lv_micropython/py/usermod.cmake:9 (get_target_property):
  get_target_property() called with non-existent target "lvgl_interface".
Call Stack (most recent call first):
  /home/le/esp/lv_micropython/py/usermod.cmake:24 (usermod_gather_sources)
  /home/le/esp/lv_micropython/py/usermod.cmake:24 (usermod_gather_sources)
  /home/le/esp/lv_micropython/py/usermod.cmake:47 (usermod_gather_sources)
  main/CMakeLists.txt:14 (include)

CMake Error at /home/le/esp/lv_micropython/py/usermod.cmake:15 (get_target_property):
  get_target_property() called with non-existent target "lvgl_interface".
Call Stack (most recent call first):
  /home/le/esp/lv_micropython/py/usermod.cmake:24 (usermod_gather_sources)
  /home/le/esp/lv_micropython/py/usermod.cmake:24 (usermod_gather_sources)
  /home/le/esp/lv_micropython/py/usermod.cmake:47 (usermod_gather_sources)
  main/CMakeLists.txt:14 (include)

CMake Error at /home/le/esp/lv_micropython/py/usermod.cmake:21 (get_target_property):
  get_target_property() called with non-existent target "lvgl_interface".
Call Stack (most recent call first):
  /home/le/esp/lv_micropython/py/usermod.cmake:24 (usermod_gather_sources)
  /home/le/esp/lv_micropython/py/usermod.cmake:24 (usermod_gather_sources)
  /home/le/esp/lv_micropython/py/usermod.cmake:47 (usermod_gather_sources)
  main/CMakeLists.txt:14 (include)

PGNetHun, 2024-02-16:
Hello!
Current MicroPython-LVGL binding version (MP: v1.20, LVGL: v9) does not support ESP32-S3, but I'm going to upgrade to latest MicroPython version (1.22.x), it just needs time.

Btw there is an another repo with MicroPython v1.22 & LVGL v9 you can check: https://github.com/kdschlosser/lvgl_micropython
The main difference is that there the LVGL is added to MicroPython as a separate user module.

kk8900, 2024-02-19:
> Hello! Current MicroPython-LVGL binding version (MP: v1.20, LVGL: v9) does not support ESP32-S3, but I'm going to upgrade to latest MicroPython version (1.22.x), it just needs time.
>
> Btw there is an [...]

Thank you for your reply.

PGNetHun, 2025-03-15:
Hello @kk8900 
lv_micropython is updated to MicroPython v1.24.1, so you should be able to build ESP32-S3 firmware.

Please install ESP-IDF v5.2.3 to build ESP32 port: 
https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32s3/get-started/linux-macos-setup.html

I've added also [build scripts](https://github.com/PGNetHun/lv_micropython/tree/master/scripts), so you just need to run `build-esp32.sh` script, and select the correct board & variant, then run `deploy-esp32.sh` script to deploy to your device.

kk8900, 2025-03-17:
> Hello [@kk8900](https://github.com/kk8900) `lv_micropython` is updated to MicroPython v1.24.1, so you should be able to build ESP32-S3 firmware.
>
> Please install ESP-IDF v5.2.3 to build ESP32 port: [...]

Thank you. I will try it when I have time.

kk8900, 2025-03-18:
> Hello [@kk8900](https://github.com/kk8900) `lv_micropython` is updated to MicroPython v1.24.1, so you should be able to build ESP32-S3 firmware.
>
> Please install ESP-IDF v5.2.3 to build ESP32 port: [...]

Hello,

Thank you for the update and the build scripts you provided. They have made the entire process much simpler.

However, I encountered a problem. By default, the firmware does not include support for display and touch drivers. When I modified `lv_conf.h` to enable the display driver, an error occurred. Here is the error message:

[1326/2108] Building C object esp-idf/...__idf_main_esp32.dir/__/__/lv_mp.c.obj
FAILED: esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/__/__/lv_mp.c.obj 
/home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/lv_mp.c: In function 'mp_lv_lcd_generic_mipi_create':
/home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/lv_mp.c:43640:122: error: invalid type argument of '->' (have 'uint32_t' {aka 'long unsigned int'})
43640 |     void *send_cmd_cb = mp_lv_callback(mp_args[3], &uint32_t_send_cmd_cb_callback, MP_QSTR_uint32_t_send_cmd_cb, &hor_res->user_data, NULL, (mp_lv_get_user_data)NULL, (mp_lv_set_user_data)NULL);
      |                                                                                                                          ^~
/home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/lv_mp.c:43641:128: error: invalid type argument of '->' (have 'uint32_t' {aka 'long unsigned int'})
43641 |     void *send_color_cb = mp_lv_callback(mp_args[4], &uint32_t_send_color_cb_callback, MP_QSTR_uint32_t_send_color_cb, &hor_res->user_data, NULL, (mp_lv_get_user_data)NULL, (mp_lv_set_user_data)NULL);
      |                                                                                                                                ^~
/home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/lv_mp.c: In function 'mp_lv_ili9341_create':
/home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/lv_mp.c:43741:122: error: invalid type argument of '->' (have 'uint32_t' {aka 'long unsigned int'})
43741 |     void *send_cmd_cb = mp_lv_callback(mp_args[3], &uint32_t_send_cmd_cb_callback, MP_QSTR_uint32_t_send_cmd_cb, &hor_res->user_data, NULL, (mp_lv_get_user_data)NULL, (mp_lv_set_user_data)NULL);
      |                                                                                                                          ^~
/home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/lv_mp.c:43742:128: error: invalid type argument of '->' (have 'uint32_t' {aka 'long unsigned int'})
43742 |     void *send_color_cb = mp_lv_callback(mp_args[4], &uint32_t_send_color_cb_callback, MP_QSTR_uint32_t_send_color_cb, &hor_res->user_data, NULL, (mp_lv_get_user_data)NULL, (mp_lv_set_user_data)NULL);
      |                                                                                                                                ^~
/home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/lv_mp.c: In function 'mp_lv_st7789_create':
/home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/lv_mp.c:43778:122: error: invalid type argument of '->' (have 'uint32_t' {aka 'long unsigned int'})
43778 |     void *send_cmd_cb = mp_lv_callback(mp_args[3], &uint32_t_send_cmd_cb_callback, MP_QSTR_uint32_t_send_cmd_cb, &hor_res->user_data, NULL, (mp_lv_get_user_data)NULL, (mp_lv_set_user_data)NULL);
      |                                                                                                                          ^~
/home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/lv_mp.c:43779:128: error: invalid type argument of '->' (have 'uint32_t' {aka 'long unsigned int'})
43779 |     void *send_color_cb = mp_lv_callback(mp_args[4], &uint32_t_send_color_cb_callback, MP_QSTR_uint32_t_send_color_cb, &hor_res->user_data, NULL, (mp_lv_get_user_data)NULL, (mp_lv_set_user_data)NULL);
      |                                                                                                                                ^~
[1328/2108] Building C object esp-idf/...opython/lvgl/src/core/lv_obj_pos.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stderr_output_67497 and /home/le/esp/lv_micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stdout_output_67497
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:71：all] 错误 1

Could you provide some suggestions on how to resolve this issue? I really need your help.

Additionally, the MicroPython version is showing as `MicroPython v1.20.0-2503.g0672cd8c5`, which seems incorrect.

Best regards,

PGNetHun, 2025-03-18:
Hello @kk8900 ,
Yes, I know about that problem, at the moment there is no fix for that, but after Github workflow CI builds are fixed, I will continue to work on display driver support (using these LVGL built-in drivers instead of ESP32->ILI9xxx.py driver).

So till it's fixed, please use old lv_micropython (before changes commit): https://github.com/lvgl/lv_micropython/commit/bf110742007ed2519e0d688aa58ef5c2f0acfaf0

PS:
thanks for pointing that version issue, I add that to my TODO list :)

kk8900, 2025-03-25:
> Hello [@kk8900](https://github.com/kk8900) , Yes, I know about that problem, at the moment there is no fix for that, but after Github workflow CI builds are fixed, I will continue to work on display driver support (using these LVGL built-in drivers instead of binding's ESP32->ILI9xxx.py driver).
> 
> [#341 (comment)](https://github.com/lvgl/lv_binding_micropython/pull/341#issuecomment-2731069902)
> 
> So till it's fixed, please use old `lv_micropython` (before changes commit): https://github.com/lvgl/lv_micropython/tree/bf110742007ed2519e0d688aa58ef5c2f0acfaf0
> 
> PS: thanks for pointing that version issue, I add it to my TODO list :)

Hello, PGNetHun. How can I compile other drivers into the lv_micropython firmware?

I used the command `make USER_C_MODULES=~/esp/micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT` to try to compile the [micropython-camera-API](https://github.com/cnadler86/micropython-camera-API.git) into the firmware. The firmware compiles successfully, but it does not include the camera driver. This method is effective when compiling the micropython firmware, but it is not effective when compiling the lv_micropython firmware.