lvgl/lv_binding_micropython #337: Ask a question about buttons  of the button

xulei3388, 2024-03-13:
Ask a question about buttons

Manually set the current status of the button

How to use the lv_btn_set_state instruction in the mpy+lvgl system?

PGNetHun, 2024-03-19:
Hello!
Every `lv_` function can be divided into 2 main groups in MicroPython code:
1. Class methods: can be used independent from any LVGL object instance, or enums, constants, constructors, etc. For example: `lv.button()` = create new button, `lv.obj.FLAG.HIDDEN` = object hidden flag
2. Instance methods: after LVGL object is created, we can call methods bound to it. For example: lbl.set_text() = set `lv.label` object's text, `btn.set_state()` = set a button object's state

Basically wherever you see `lv_xxx_yyyy()` widget functions in [LVGL documentation](https://docs.lvgl.io/master/widgets/index.html), you can "translate" those to: `xxx.yyyy()`.

In your case:
there is no "set_state" function, but `add_state` and `remove_state` [on the object](https://docs.lvgl.io/master/overview/obj.html#states):
- C code: `lv_obj_add_state(btn1, LV_STATE_DISABLED)`
- MP code: `btn1.add_state(lv.STATE.DISABLED)`

xulei3388, 2024-03-22:
Best regards, thank you very much!

Your selfless answer!

closed 2024-03-25 by PGNetHun