lvgl/lv_binding_micropython #348: touch ft6x36 stops working

pask81, 2024-07-03:
I am using rpi pico with generic driver ili9xxx for the screen and ft6x36 for the touch. Everything works well normally, but randomly after some time running the touchscreen stops responding, while the rest of the application continues to work. Are there any know reasons for ft6x36 to show this problem, or can anyone give tips to solve or at least debug the problem?

Octopus1633, 2024-12-28:
@pask81 Hello,Have you solved this problem? I am also facing this problem. It can work stably before adding touch. After adding touch, there is a probability that lvgl will stop working when touching, but the main loop is still running. This is the error given by the program:
```
Traceback (most recent call last):
  File "/lib/lv_utils.py", line 124, in task_handler
KeyError: lv_display_t_flush_cb
```

pask81, 2025-01-09:
> @pask81 Hello,Have you solved this problem? I am also facing this problem. It can work stably before adding touch. After adding touch, there is a probability that lvgl will stop working when [...]

My problem was my program was updating graphical element in response to events different from touches, but lvgl is not thread-safe. It is safe to call LVGL functions in
 event callbacks (i.e. reactions to touches) or 
timer callbacks, so now when another event needs to update a graphic element, it changes a status variable that is checked in a timer event and the callback of the timer event actually update the graphic element. 

This mostly solved the problem, but in some cases I still got a fozen touchscreen... To avoid having a dead device in these cases I used the watchdog to force a reboot when the event loop is not running anymore. Here is a snippet of my code for this

import machine
from lv_utils import event_loop

wdt = machine.WDT(timeout=8388)

async def wdtd():
    while event_loop.is_running():
        wdt.feed()
        await uasyncio.sleep(1)