lvgl/lv_binding_micropython #353: building lv_micropython on windows

jonhp, 2024-09-19:
Are the lv_micropython build instructions meant to be followed on windows 11?
Because I can’t get them to work…
Or do the have to be run on linux; in which case perhaps a warning should be
Added to the readme.md.
Thanks, jon

kisvegabor, 2024-09-23:
cc @kdschlosser @liamHowatt @amirgon

amirgon, 2024-09-23:
See also https://github.com/lvgl/lv_binding_micropython/issues/175

kdschlosser, 2024-09-23:
It depends on what you are trying to compile. If you are compiling for the ESP32 and you are on a Windows desktop PC then you will need to use WSL in order to do that. If you are wanting to compile MicroPython as a native windows application then you would need to read the link that @amirgon posted above. With Windows 11 using WSL2 you can compile for unix using WSL2 and it will give you a proper Window as if it was being natively run on Windows. You can even create a shortcut to start the application and it will work.