lvgl/lv_binding_micropython #361: lvgl v9.2.0 support broken.

andrewleech, 2024-11-13:
I've attempted to build against LVGL v9.2.0 and while it compiles, I get an error thrown at runtime from code like:
``` python

def HomeBtnBlockB_eventhandler(event_struct):
   event = event_struct.code
   if event == lv.EVENT.CLICKED and True:
      print("clicked")

ui_HomeBtnBlockB = lv.imagebutton(ui_Home)
ui_HomeBtnBlockB.add_event_cb(HomeBtnBlockB_eventhandler, lv.EVENT.ALL, None)
```

The `add_event_cb()` function throws:
```
Traceback (most recent call last):
  File "app.py", line 23, in <module>
  File "lvgl/demo1/export/ui.py", line 227, in <module>
SyntaxError: Cannot convert 'function' to pointer!
```

In the generated `lv_mpy.c` the error is thrown by:
[image]

`event_cb = mp_to_ptr()` won't accept a function, the exception is thrown from https://github.com/lvgl/lv_binding_micropython/blob/96d6d3538fec4de99c200c1bb388b02b95cbbdaf/gen/gen_mpy.py#L1355

I've bisected lvgl and narrowed down this change in behavior to https://github.com/lvgl/lvgl/commit/1d14386b
Before this it works fine.

Carglglz, 2024-12-05:
See https://github.com/lvgl/lvgl/blob/master/docs/CHANGELOG.rst#breaking-changes, I had similar problems but I've just updated `lv_conf.h` in my branch https://github.com/lvgl/lv_binding_micropython/pull/341/commits/6a5eb8810d76730097b760f1036e8218ee0ae3a5 and it seems to work now 👍🏼 (cc @Lzw655)

Lzw655, 2024-12-11:
Hi @andrewleech @Carglglz, thank you very much for your efforts! I tested the branch https://github.com/lvgl/lv_binding_micropython/commit/6a5eb8810d76730097b760f1036e8218ee0ae3a5, and it indeed resolves this issue. 👍

andrewleech, 2025-02-19:
I've integrated the fix into [#242](https://github.com/lvgl/lv_binding_micropython/pull/242) as well so consider this resolved thanks.

closed 2025-02-19 by andrewleech