lvgl/lv_binding_micropython #372: missing espidf MicroPython module

selcukgun, 2025-03-21:
I'm encountering an issue where the espidf MicroPython module is not generated during the normal build process documented in [lv_micropython repo](https://github.com/lvgl/lv_micropython/blob/master/README-LVGL.md).

**Steps to Reproduce:**
1. Run the following command:
`make -C ports/esp32 BOARD=ESP32_GENERIC_C6 LV_CFLAGS="-DLV_COLOR_DEPTH=16" PORT=/dev/ttyACM0 deploy`
2. Attempt to import the module in MicroPython:
`from ili9XXX import st7789`
3. This results in:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ili9XXX.py", line 19, in <module>
ImportError: no module named 'espidf'
```
**Additional Context:**

I attempted to generate the bindings using the gen_mpy.py tool manually by passing hundreds of include flags, but I encountered dependency issues (e.g., missing headers) and the process still failed to produce the complete module.

It appears that the espidf module is not being generated as expected. is there any other way to use st7789 display?

**Request:**

1. Is there a build option or flag within the default build system that ensures the espidf MicroPython module is generated?
2. Alternatively, can the generation of the espidf module be integrated into the standard build process (i.e., via the make command above) without needing to manually invoke gen_mpy.py?
3. 
Any guidance or fixes would be greatly appreciated, as manual generation with gen_mpy.py has proven to be error-prone due to the large number of dependencies.

kdschlosser, 2025-03-25:
It longer longer works with the ESP32.. I am not sure why all of the esp32 related build stuff was removed but it was.

PGNetHun, 2025-03-25:
Hello!

This is a known issue, the file `espidf.c` is currently not compiled into an MicroPython module, as it needs to be updated.
The issue is that `CONFIG_IDF_TARGET_*` is not defined during compilation, and [esp32/clk.h](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/espidf.h#L115C14-L115C25) is not found.
This is on my list to fix.
Till then it is not possible to use drivers from `ili9xxx.py`.
My plan is also to include [LVGL builtin display drivers](https://github.com/lvgl/lvgl/tree/master/src/drivers/display) instead of this ili9xxx.py, but `gen_mpy.py` generator has a bug when those are included in build process.

So, please patient till `espidf.c` issue is fixed, or go with older `lv_micropython` version.

celiu-2000, 2025-03-29:
I have also encountered the same problem. When can it be solved?

selcukgun, 2025-03-30:
I gave a try to this project:  https://github.com/lvgl-micropython/lvgl_micropython

The following one liner worked great. 

```bash
python3 make.py esp32 BOARD=ESP32_GENERIC_C6 DISPLAY=st7789
````

it will print esptool commad at the end like the following:

```bash
/home/selcuk/.espressif/python_env/idf5.2_py3.10_env/bin/python -m esptool --chip esp32c6 -p /dev/ttyACM0 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 4MB --flash_freq 80m --erase-all 0x0 /home/selcuk/projects/esp32/lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_C6-4.bin
```

After this you can use the sample code for LVGL in the repo after setting your pins for SPI etc.

@kdschlosser Thank you for this smooth solution!

kdschlosser, 2026-01-11:
The project appears as tho it is not really being actively maintained. You might want to try out [lvgl_micropython](https://github.com/lvgl-micropython/lvgl_micropython). it does support the ESP32 and it comes packaged with 36 display drivers (plus variations), 17 touch drivers, 9 io expander drivers, with SPI, Quad SPI, Octal SPI, 8 lane I8080, 16 lane I8080, 8 lane RGB, 16 lane RGB, I2C and MIPI DSI connections to the display.

Very simply one line build command and you don't have to worry about collecting any dependencies it does it for you. MicroPython v1.27.0, ESP-IDF v5.5.1 and LVGL v9.4.0