lvgl/lv_binding_micropython #373: Can't get indev point

bhcuong2008, 2025-03-25:
Hi,

I use the latest of lv_binding_micropython using LVGL v9.2.0-624-g9c0431676. I dont know how to get indev point after many search on the Internet. It has `lv.indev_t.get_point()` with 2 arguments. The 2nd arg of point_t is ok, I could create with `p = lv.point_t()`. But the 1st arg is indev_t, I dont know what to pass to it. I tried `lv.indev_active()` as follow but not success with error `TypeError: argument has wrong type`.

`lv.indev_t.get_point(lv.indev_active(), p)`

In C program, I could read indev point data via read_cb passed to `lv_indev_set_read_cb` as below:
```
LV_INDEV_STATE_PRESSED, x: 292 | y: 161
LV_INDEV_STATE_PRESSED, x: 292 | y: 161
LV_INDEV_STATE_PRESSED, x: 292 | y: 161
```

bhcuong2008, 2025-03-25:
I think it may be wrong with generated 
`static MP_DEFINE_CONST_LV_FUN_OBJ_STATIC_VAR(mp_lv_indev_get_active_obj_mpobj, 0, mp_lv_screen_active, lv_indev_get_active_obj);`

It should use `mp_lv_indev_create`, not `mp_lv_screen_active`

bhcuong2008, 2025-03-25:
I also tried, `indev = lv.indev_t.get_next(None)` but with error `TypeError: argument has wrong type`

kdschlosser, 2025-03-25:
```python
indev = lv.indev_active()

p = lv.point_t()
indev.get_point(p)

print(p.x, p.y)
```

kdschlosser, 2025-03-25:
```python
indev = lv.indev_active()
p = lv.point_t()

while indev:
    indev.get_point(p)
    print(p.x, p.y)
    indev = indev.get_next()
```

bhcuong2008, 2025-03-25:
Thank u for ur guide. At my side, it returns NoneType, not struct lv_indev_t
FYI, I init my indev in C side, not in MicroPython side. So is it ok?

```
>>> indev = lv.indev_active()
>>> 
>>> type(indev)
<class 'NoneType'>
```

bhcuong2008, 2025-03-25:
> I also tried, `indev = lv.indev_t.get_next(None)` but with error `TypeError: argument has wrong type`

I think get_next function should support None arg like NULL in C side, `lv_indev_t *lv_indev_get_next(lv_indev_t *indev)`, so that it will return the 1st indev.

bhcuong2008, 2025-03-25:
In generated lv_mp.c, function `static mp_obj_t mp_lv_indev_get_next(size_t mp_n_args, const mp_obj_t *mp_args, void *lv_func_ptr)` does not check argument type. But in REPL, when I pass None, `lv.indev_t.get_next(None)`, it reports `TypeError: argument has wrong type`.

I tried to find but not found any code rejecting None type and other types too. It only accepts lv_indev_t type.

bhcuong2008, 2025-03-25:
I could do it. Due to get_next is member method of class lv_indev_t, so MPY checks 1st arg of any method to make sure that it is type of lv_indev_t. MPY has option MICROPY_BUILTIN_METHOD_CHECK_SELF_ARG with default (1) to check self arg. 

I disable that option in mpconfigport.h by `#define MICROPY_BUILTIN_METHOD_CHECK_SELF_ARG (0)` then I could call `indev = lv.indev_t.get_next(None)` and get point with `indev.get_point(p)`.

closed 2025-03-25 by bhcuong2008