lvgl/lv_binding_micropython #381: Many 404 error in the online simulator

kisvegabor, 2025-05-05:
Hi,

It seems the recent updates introduced a lot of "file not found" issues. See the developer console at https://sim.lvgl.io/v9.0/micropython/ports/webassembly/index.html

[image]

@PGNetHun  
Does it say anything to you?

Carglglz, 2025-05-08:
@kisvegabor I don't think the recent updates are related to these errors, if I'm not wrong the simulator is built from [`lv_micropython/lvgl_javascript_v9`](https://github.com/lvgl/lv_micropython/tree/lvgl_javascript_v9) branch, see last commit hash [349ae32bf](https://github.com/lvgl/lv_micropython/commit/349ae32bfb91b8d7d259907ca80fb1b55617c131) and MicroPython REPL: 
```
Running the SDL lvgl version

MicroPython v1.20.0-1053-g349ae32bf-dirty on 2024-09-11; JS with Emscripten
Type "help()" for more information.
>>> 
```

Errors come from [`lv_micropython/ports/webassembly/modules/imp.py`](https://github.com/lvgl/lv_micropython/blob/349ae32bfb91b8d7d259907ca80fb1b55617c131/ports/webassembly/modules/imp.py#L44) which import files from `lv_binding_micropython` repo, so I think there are two "issues" here: 

1) `print` function output is redirected to `sys.stderr`, so normal prints will appear as errors

2) The import function iterates over `sys.path` so e.g. it will try to import `lv_utils.py` from `drivers/js` which is not the correct path and therefore it will get a `404 (Not Found)`

Probably this should be improved or documented in some way because right now it is quite confusing.

kisvegabor, 2025-05-09:
Ah, thank you for pointing that out. 👍 
I close this issue now, but let's keep this improvement possibility in mind.